prusa3d/KiKIt_fab_prusa #8: Přibalit KiKit k Prusamanovi

yaqwsx, 2022-05-11:
Jelikož je vývoj provázán, dává smysl, aby instalace nového Prusamana aktualizovala KiKit. Preferovaná varianta je přibalovat KiKit a instalovat oba doplňky.

ok1hra, 2022-05-11:
Potrebe aktualizace KiKitu rozumim, nicmene - neznamena pribaleni KiKit do instalace Prusamana moznost jeho zastarani v pripade, ze nebude potreba Prusamana dale vyvijet?

yaqwsx, 2022-05-15:
To je něco, nad čím dává smysl se v budoucnu ještě zamyslet - a je to spíše filozofická otázka (technicky je možné oboje):
- je možné se na to dívat tak, že Prusaman a KiKit pro vaše potřeby tvoří jeden balíček (který je +- zakonzervovaný a plní vaše potřeby). Pak přibalení dává smysl - zejména pokud KiKit někdy diverguje, tak tato varianta nezpůsobí nefunkčnost Prusamana.
- samotná případná aktualizace (v případě opravy chyby v KiKitu) je velmi jednoduchá - stačí ho v podstatě znovu zkompilovat a vydat novou verzi.
- Stále cílím na tom, že provázanost KiKitu a Prusamana je minimální a Prusaman spoléhá na stabilní rozhraní KiKitu. Tím pádem, v momentě, kdy si řekneme, že "takto to stačí", můžeme přestat přibalovat KiKit a držet upstream verzi.

yaqwsx, 2022-05-15:
Přibalování implementováno v f1d7d703879c31843381e987cdf5d2ecd274181e.

closed 2022-05-15 by yaqwsx