prusa3d/KiKIt_fab_prusa #20: Zrušit Prusaman YAML

yaqwsx, 2022-05-16:
Původní `prusaman.yaml` existovat protože bylo třeba:
- znát ID a revizi desky
- panelizační parametry
- mělo se za to, že se má Prusaman nasadit jednoduše na starší desky a tudíž měl podporovat více filtrů.

Avšak:
- panelizační parametry mají vlastní soubor `kikit.json` a zatím není viděna potřeba jiného stylu panelizace (pokud bude, může se Prusaman rozhodovat na základě existence `kikit.json`, `panel.kicad_pcb` (manuální panel), `panel.sh` (panelizační skript).
- v procesu výroby se stejně přepisuje a anotuje schéma, takže nedává smysl podporovat starý hvězdičkový filter.
- ID a revize desky se nachází ve zdrojových datech.

Tudíž, `prusaman.yaml` není třeba a místo toho:
- ID desky si přečteme ve schématu
- Revizi v PCB
- panelizuje se podle existujících souborů
- a používá se pouze PnB filter.

ok1hra, 2022-05-16:
Souhlasim
- master pro verzi je nastaveni v kicadu (stejne napriklad pro generovane pdf schema)
- protoze nastaveni verze muze byt jine ve schematu a pcb, bylo by vhodne doplnit kontrolou, ze jsou obe hodnoty stejne
- panelizujeme existujici soubory a pouzivame vyhradne PnB filtr

[image]

yaqwsx, 2022-05-16:
Mohu jen poprosit o dopřesnění, které shody je třeba kontrolovat? Je to to, že pole `Komentář 4` je rovno `PCB ID-${ID}`?

ok1hra, 2022-05-16:
Kontrolovat, revize v Nastaveni stranky ve schematu je shodna s revizi v Nastaveni stranky ve pcb.
Ty ID by mely byt pres promennou globalne stejne.
Diky

yaqwsx, 2022-05-18:
Implementováno v 990b3a7