prusa3d/KiKIt_fab_prusa #25: Nalezeni minimal drilling/guide width

ok1hra, 2022-05-18:
- [ ] Nalezt nejmensi vrtani na desce a doplnit tento udaj do readme
Stavajici format:
`- minimal drilling of {min-drill} mm - hole diameters are trailing - surface treatment immersive gold`
Soucasny stav je vice pranim, nez odrazem skutecnosti.

- [ ] stejna situace je u minimalni sirky spoje, coz nevim zda je implementovatelne
`- Minimum guide width / 0.2 mm gaps`

yaqwsx, 2022-05-18:
Bod je 1 je bez problémů implementovatelný. U druhého bodu je problém s mezerou; napadají mě různá řešení:
- použít údaj z parametrů desky. Pokud deska prochází DRC (a to budeme kontrolovat), tak je údaj pravdivý.
- pokud je třeba znát skutečně minimální mezeru, tak bychom mohli neužít DRC a správnou hodnotu nalézt binárním vyhledáváním.

ok1hra, 2022-05-18:
Pouzit udaj z parametru desky muzeme, ale implementovat az budeme kontrolovat DRC, jinak by mohl byt udaj nespravny.
Pokud ti prijde druha varianta schudnejsi, nejsem proti ;)

yaqwsx, 2022-05-18:
Druhá variantu můžeme vyzkoušet; uvidíme, jak moc bude brzdit. Odhaduji, že bude znamenat zavolat cca 10 kontrolu DRC.

yaqwsx, 2022-05-25:
Po několika experimentech jsem dospěl k tomu, že bude lepší kontrolovat DRC + číst hodnoty z návrhových pravidel. Kdyby bylo skutečně potřeba hledat nejmenší hodnoty, které deska má, tak můžeme předělat (jen mi teď nepřipadá, že by efekt stál za vynaložené úsilí).

Implementováno v 5039007

closed 2022-05-25 by yaqwsx