prusa3d/KiKIt_fab_prusa #28: Prava souboru

ok1hra, 2022-05-19:
Na linuxu pri pouziti sdileneho sitoveho disku maji vystupni soubory Prusamana omezena prava.
Kolegove pak nemaji na sitovem disku k souborum pistup.
[image]
Je mozne zmenit na -rw-rw-rw?

Stejny problem se tyka i samotnych zdrojovych souboru kicadu, kde je to jeste striktnejsi
[image]
coz nevim zda je v ramci prusamana resitelne...

yaqwsx, 2022-05-19:
Toto bude způsobené kombinací toho, jak generuji výstupy:
- založím dočasný adresář, do kterého nageneruji všechny výstupy (proto, abych v případě chyby neničil existující výstup)
- následně provedu prohození cílového s tímto dočasným.
- 
Což povede k tomu, že cílový adresář bude mít přístupová práva stejná jako temp lokace. Můžu ale místo dočasného adresáře použít adresář v cílové lokaci a tím pádem dostat implicitní práva dané lokace.

yaqwsx, 2022-05-19:
ba5c4ed by to mohlo řešit, ale nemám to u sebe jak otestovat.

closed 2022-07-04 by ok1hra