prusa3d/KiKIt_fab_prusa #61: 'charmap' codec can't decode byte 0x8d

dzarda, 2022-09-16:
Windows 10 64bit
Kicad 6.0.7
Prusaman 0.2.0+121

Zřejmě čtení zdrojových dat nezvládne něco požvýkat. Budeš potřebovat zdroj?

[image]

```
Traceback (most recent call last):
  File "C:\Users\Dzarda\Documents\KiCad\6.0\3rdparty\Python39\site-packages\prusaman\gui\make.py", line 120, in doExportWork
    raise exception
  File "C:\Users\Dzarda\Documents\KiCad\6.0\3rdparty\Python39\site-packages\prusaman\gui\make.py", line 108, in work
    generator.make()
  File "C:\Users\Dzarda\Documents\KiCad\6.0\3rdparty\Python39\site-packages\prusaman\manugenerator\__init__.py", line 114, in make
    self._makeBoardStage()
  File "C:\Users\Dzarda\Documents\KiCad\6.0\3rdparty\Python39\site-packages\prusaman\manugenerator\boardStage.py", line 33, in _makeBoardStage
    self._makeIbom(source=outfile, outdir=outdir)
  File "C:\Users\Dzarda\Documents\KiCad\6.0\3rdparty\Python39\site-packages\prusaman\manugenerator\__init__.py", line 191, in _makeIbom
    bom = extractComponents(str(self._project.getSchema()))
  File "C:\Users\Dzarda\Documents\KiCad\6.0\3rdparty\Python39\site-packages\kikit\eeschema_v6.py", line 148, in extractComponents
    symbols, instances = collectSymbols(filename)
  File "C:\Users\Dzarda\Documents\KiCad\6.0\3rdparty\Python39\site-packages\kikit\eeschema_v6.py", line 125, in collectSymbols
    s, i = collectSymbols(f, path + "/" + uuid)
  File "C:\Users\Dzarda\Documents\KiCad\6.0\3rdparty\Python39\site-packages\kikit\eeschema_v6.py", line 113, in collectSymbols
    sheetSExpr = parseSexprF(f)
  File "C:\Users\Dzarda\Documents\KiCad\6.0\3rdparty\Python39\site-packages\kikit\sexpr.py", line 183, in parseSexprF
    lw = readWhitespace(stream)
  File "C:\Users\Dzarda\Documents\KiCad\6.0\3rdparty\Python39\site-packages\kikit\sexpr.py", line 138, in readWhitespace
    c = stream.peek()
  File "C:\Users\Dzarda\Documents\KiCad\6.0\3rdparty\Python39\site-packages\kikit\sexpr.py", line 27, in peek
    self.pending = self.stream.read(1)
  File "C:\Program Files\KiCad\6.0\bin\lib\encodings\cp1252.py", line 23, in decode
    return codecs.charmap_decode(input,self.errors,decoding_table)[0]
UnicodeDecodeError: 'charmap' codec can't decode byte 0x8d in position 576: character maps to <undefined>
```

dzarda, 2022-09-19:
Bohužel stejná chyba na verzi +124

yaqwsx, 2022-09-20:
Resolved in [fe06c2a](https://github.com/prusa3d/KiKIt_fab_prusa/commit/fe06c2addd957c5b863b67bd0decd8378b14e4c3)

closed 2022-09-20 by yaqwsx